LuckfoxTECH/luckfox-pico #235: custom camera usage

niyazFattahov, 2025-02-05:
I was trying to use RPI ov5647 camera.
This customized dts 'rv1106-luckfox-pico-pro-max-ipc.dtsi' (where I added ov5647):

```
#include "rv1106-amp.dtsi"

/ {
	chosen {
		bootargs = "earlycon=uart8250,mmio32,0xff4c0000 console=ttyFIQ0 root=/dev/mmcblk1p7 rootwait snd_soc_core.prealloc_buffer_size_kbytes=16 coherent_pool=0";
	};

	acodec_sound: acodec-sound {
		compatible = "simple-audio-card";
		simple-audio-card,name = "rv-acodec";
		simple-audio-card,format = "i2s";
		simple-audio-card,mclk-fs = <256>;
		simple-audio-card,cpu {
			sound-dai = <&i2s0_8ch>;
		};
		simple-audio-card,codec {
			sound-dai = <&acodec>;
		};
	};

	vcc_1v8: vcc-1v8 {
		compatible = "regulator-fixed";
		regulator-name = "vcc_1v8";
		regulator-always-on;
		regulator-boot-on;
		regulator-min-microvolt = <1800000>;
		regulator-max-microvolt = <1800000>;
	};

	vcc_3v3: vcc-3v3 {
		compatible = "regulator-fixed";
		regulator-name = "vcc_3v3";
		regulator-always-on;
		regulator-boot-on;
		regulator-min-microvolt = <3300000>;
		regulator-max-microvolt = <3300000>;
	};

	vdd_arm: vdd-arm {
		compatible = "regulator-fixed";
		regulator-name = "vdd_arm";
		regulator-min-microvolt = <800000>;
		regulator-max-microvolt = <1000000>;
		regulator-init-microvolt = <900000>;
		regulator-always-on;
		regulator-boot-on;
	};
	leds: leds {
		compatible = "gpio-leds";
		work_led: work{
			gpios = <&gpio3 RK_PC6 GPIO_ACTIVE_HIGH>;
			linux,default-trigger = "activity";
			default-state = "on";
		};	
	};

	// DHT11
	dht11_sensor {
        compatible = "dht11";
        pinctrl-names = "default";
        pinctrl-0 = <&gpio1_pc7>;

        dht11@1 {
            gpios = <&gpio1 RK_PC7 GPIO_ACTIVE_HIGH>;
            label = "dht11";
            linux,default-trigger = "humidity";
[... 557 lines cut ...]
[    3.106613] dfa0:                   00000000 00000001 00000001 00522cb8 00000002 00000000
[    3.114787] dfc0: 00000000 00000001 00522cb8 00000004 005215b4 a6febec4 ae834780 00000020
[    3.122958] dfe0: 00520c38 ae834748 0044e238 a6f5b670
[    3.128013] Code: f8d0 3588 7804 6845 (681a) f243 
[    3.132851] ---[ end trace 6e763bd7641e803e ]---
[    3.375325] rk_gmac-dwmac ffa80000.ethernet eth0: PHY [stmmac-0:02] driver [RK630 PHY] (irq=POLL)
[    3.384509] dwmac4: Master AXI performs any burst length
[    3.389833] rk_gmac-dwmac ffa80000.ethernet eth0: No Safety Features support found
[    3.397428] rk_gmac-dwmac ffa80000.ethernet eth0: PTP not supported by HW
[    3.404508] rk_gmac-dwmac ffa80000.ethernet eth0: configuring for phy/rmii link mode
[    3.733799] file system registered
[    3.752083] using random self ethernet address
[    3.756540] using random host ethernet address
[    3.772269] Mass Storage Function, version: 2009/09/11
[    3.777427] LUN: removable file: (no medium)
[    3.843947] read descriptors
[    3.846855] read strings
[    4.838826] usb0: HOST MAC 52:b5:16:0e:08:3d
[    4.843145] usb0: MAC 4a:ea:c1:9d:48:00
[    5.238917] dwc3 ffb00000.usb: device reset
[    5.358537] dwc3 ffb00000.usb: device reset
[    5.442742] android_work: sent uevent USB_STATE=CONNECTED
[    5.454360] android_work: sent uevent USB_STATE=CONFIGURED
[    5.522015] rk_gmac-dwmac ffa80000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
```

luckfox-eng29, 2025-02-24:
Hello, please modify
```
			csi_dphy_input3: endpoint@3 {
				reg = <2>;
				remote-endpoint = <&ov5647_out>;
				data-lanes = <1 2>;
			};
```
to
```
			csi_dphy_input3: endpoint@3 {
				reg = <3>;
				remote-endpoint = <&ov5647_out>;
				data-lanes = <1 2>;
			};
```

niyazFattahov, 2025-02-25:
@luckfox-eng29, Hello. This changes nothing, I get same error.

luckfox-eng29, 2025-02-25:
The notable difference between your logs and a normal startup is that the initialization of components such as rkcif is performed without registering the sensor driver. If it is confirmed that this is not an issue with the device tree configuration, I need to know how you are loading the driver.
The correct process is as follows: the kernel compiles and generates the sensor's .ko file, which is then placed in /oem/usr/ko on the board. The insmod command for the sensor's .ko file must be executed before video_rkcif.ko. This can be achieved by modifying the /oem/usr/ko/insmod_ko.sh script.

niyazFattahov, 2025-02-25:
this is my '

> nsmod_ko.sh

 content:

`#!/bin/sh
cmd=`realpath $0`
_DIR=`dirname $cmd`
cd $_DIR

udevadm control --stop-exec-queue

__insmod()
{
	if [ -f "$1" ];then
		insmod $@
	fi
}

__rmmod_camera_sensor()
{
	for item in `echo "ov5647 imx415 os04a10 sc4336 sc3336 sc530ai gc2053 sc200ai sc401ai sc450ai techpoint"`
	do
		if lsmod | grep $item | awk '{print $3}' |grep -w 0;then
			rmmod $item
		fi
	done
}

__chk_camera_sensor_height()
{
	# TODO: Do Not Support dual Camera Sensor
	sensor_height=0
	for item in `echo "/proc/rkisp-vir0 /proc/rkisp0"`
	do
		if grep -w "Input.*Format" $item; then
			msg_sen=`grep -w "Input.*Format" $item`
			msg_sen=${msg_sen##*Size:*x}
			msg_sen=${msg_sen%%@*}
			sensor_height="$msg_sen"
			break
		fi
	done
}

__insmod rk_dvbm.ko

__insmod videobuf2-memops.ko
__insmod videobuf2-common.ko
__insmod videobuf2-v4l2.ko
__insmod videobuf2-vmalloc.ko
__insmod videobuf2-cma-sg.ko

__insmod ov5647.ko
__insmod imx415.ko
__insmod os04a10.ko
__insmod sc4336.ko
__insmod sc3336.ko
__insmod sc530ai.ko
__insmod gc2053.ko
__insmod sc200ai.ko
__insmod sc401ai.ko
__insmod sc450ai.ko
__insmod techpoint.ko

__insmod video_rkcif.ko
__insmod video_rkisp.ko
__insmod phy-rockchip-csi2-dphy-hw.ko
__insmod phy-rockchip-csi2-dphy.ko

__rmmod_camera_sensor

echo 1 > /sys/module/video_rkcif/parameters/clr_unready_dev
echo 1 > /sys/module/video_rkisp/parameters/clr_unready_dev

__insmod rga3.ko

__insmod mpp_vcodec.ko

__insmod rknpu.ko
__insmod snd-soc-rv1106.ko

__insmod motor.ko

__chk_camera_sensor_height

__insmod rockit.ko mcu_fw_path="./hpmcu_wrap.bin" mcu_fw_addr=0xff6fe000 isp_max_h=$sensor_height

__insmod rve.ko

udevadm control --start-exec-queue

# insmod wifi driver background
$(pwd)/insmod_wifi.sh &
`

niyazFattahov, 2025-02-25:
this is my 'insmod_ko.sh' I just added 'ov5647' before all modules:
```
#!/bin/sh
cmd=`realpath $0`
_DIR=`dirname $cmd`
cd $_DIR

udevadm control --stop-exec-queue

__insmod()
{
	if [ -f "$1" ];then
		insmod $@
	fi
}

__rmmod_camera_sensor()
{
	for item in `echo "ov5647 imx415 os04a10 sc4336 sc3336 sc530ai gc2053 sc200ai sc401ai sc450ai techpoint"`
	do
		if lsmod | grep $item | awk '{print $3}' |grep -w 0;then
			rmmod $item
		fi
	done
}

__chk_camera_sensor_height()
{
	# TODO: Do Not Support dual Camera Sensor
	sensor_height=0
	for item in `echo "/proc/rkisp-vir0 /proc/rkisp0"`
	do
		if grep -w "Input.*Format" $item; then
			msg_sen=`grep -w "Input.*Format" $item`
			msg_sen=${msg_sen##*Size:*x}
			msg_sen=${msg_sen%%@*}
			sensor_height="$msg_sen"
			break
		fi
	done
}

__insmod rk_dvbm.ko

__insmod videobuf2-memops.ko
__insmod videobuf2-common.ko
__insmod videobuf2-v4l2.ko
__insmod videobuf2-vmalloc.ko
__insmod videobuf2-cma-sg.ko

__insmod ov5647.ko
__insmod imx415.ko
__insmod os04a10.ko
__insmod sc4336.ko
__insmod sc3336.ko
__insmod sc530ai.ko
__insmod gc2053.ko
__insmod sc200ai.ko
__insmod sc401ai.ko
__insmod sc450ai.ko
__insmod techpoint.ko

__insmod video_rkcif.ko
__insmod video_rkisp.ko
__insmod phy-rockchip-csi2-dphy-hw.ko
__insmod phy-rockchip-csi2-dphy.ko

__rmmod_camera_sensor

echo 1 > /sys/module/video_rkcif/parameters/clr_unready_dev
echo 1 > /sys/module/video_rkisp/parameters/clr_unready_dev

__insmod rga3.ko

__insmod mpp_vcodec.ko

__insmod rknpu.ko
__insmod snd-soc-rv1106.ko

__insmod motor.ko

__chk_camera_sensor_height

__insmod rockit.ko mcu_fw_path="./hpmcu_wrap.bin" mcu_fw_addr=0xff6fe000 isp_max_h=$sensor_height

__insmod rve.ko

udevadm control --start-exec-queue

# insmod wifi driver background
$(pwd)/insmod_wifi.sh &

```

luckfox-eng29, 2025-02-25:
Try moving /oem/usr/ko/insmod_ko.sh to another directory to prevent it from running at startup, and then manually run insmod ov5647.ko to see what kernel log output is generated. I suspect that the issue is most likely due to the ov5647 driver failing to load, causing rkcif to be unable to successfully obtain node data, which results in the kernel error "Unable to handle kernel NULL pointer dereference at virtual address 00000000" and subsequently crashes.

niyazFattahov, 2025-02-27:
This is what I did:
mv /oem/usr/ko/insmod_ko.sh /oem/usr/ko/insmod_ko.sh.bak
reboot

I loaded manually modules:
```

insmod /oem/usr/ko/rk_dvbm.ko
insmod /oem/usr/ko/ov5647.ko
insmod /oem/usr/ko/phy-rockchip-csi2-dphy-hw.ko
insmod /oem/usr/ko/phy-rockchip-csi2-dphy.ko
insmod /oem/usr/ko/video_rkcif.ko
insmod /oem/usr/ko/video_rkisp.ko

```

```
v4l2-ctl --list-devices:
rkisp-statistics (platform: rkisp):
        /dev/video19
        /dev/video20

rkcif-mipi-lvds (platform:rkcif):
        /dev/media0

rkisp_mainpath (platform:rkisp-vir0):
        /dev/video11
        /dev/video12
        /dev/video13
        /dev/video14
        /dev/video15
        /dev/video16
        /dev/video17
        /dev/video18
        /dev/media1

Failed to open /dev/video0: No such device

```
logs:
```

Welcome to luckfox pico
luckfox login: [   55.079486] rk_dvbm ffa70000.rkdvbm: probe start
[   55.079525] rk_dvbm ffa70000.rkdvbm: dvbm ctx f17bd214
[   55.079801] rk_dvbm ffa70000.rkdvbm: probe success
[   63.368193] rockchip-csi2-dphy-hw ff3e8000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   67.112847] rockchip-csi2-dphy csi2-dphy0: csi2 dphy0 probe successfully!
[   70.025902] rkcifhw ffa10000.rkcif: no iommu attached, using non-iommu buffers
[   70.025932] rkcifhw ffa10000.rkcif: No reserved memory region assign to CIF
[   70.026311] rkcif rkcif-mipi-lvds: rkcif driver version: v00.02.00
[   70.026413] rkcif rkcif-mipi-lvds: attach to cif hw node
[   70.026429] rkcif rkcif-mipi-lvds: failed to get dphy hw node
[   70.026438] rkcif rkcif-mipi-lvds: rkcif wait line 0
[   70.026447] rkcif rkcif-mipi-lvds: rkcif fastboot reserve bufs num 3
[   70.026461] : terminal subdev does not exist
[   70.026471] : terminal subdev does not exist
[   70.026478] : terminal subdev does not exist
[   70.026486] : terminal subdev does not exist
[   70.026495] : get_remote_sensor: video pad[0] is null
[   70.026504] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[   70.026515] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[   70.026525] : get_remote_sensor: video pad[0] is null
[   70.026531] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[   70.026539] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[   70.026546] : get_remote_sensor: video pad[0] is null
[   70.026553] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[   70.026561] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[   70.026568] : get_remote_sensor: video pad[0] is null
[   70.026575] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[... 43 lines cut ...]
[   87.840673] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[   87.840681] rkcif_scale_ch0: update sensor info failed -19
[   87.840699] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.840707] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[1] get remote sensor_sd failed!
[   87.840714] rkcif_scale_ch1: update sensor info failed -19
[   87.840731] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.840739] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[2] get remote sensor_sd failed!
[   87.840745] rkcif_scale_ch2: update sensor info failed -19
[   87.840764] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.840773] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[3] get remote sensor_sd failed!
[   87.840780] rkcif_scale_ch3: update sensor info failed -19
[   87.840798] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.840805] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[   87.840813] rkcif_tools_id0: update sensor info failed -19
[   87.840830] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.840841] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[1] get remote sensor_sd failed!
[   87.840848] rkcif_tools_id1: update sensor info failed -19
[   87.840869] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.840877] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[2] get remote sensor_sd failed!
[   87.840885] rkcif_tools_id2: update sensor info failed -19
[   87.858497] rkcif-mipi-lvds: get_remote_sensor: remote pad is null
[   87.858520] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[   87.858528] stream_cif_mipi_id0: update sensor info failed -19

```

conner-replogle, 2025-03-03:
@niyazFattahov How are you connecting the camera to the board I thought the CSI connector was different to all the Raspberry Pi cameras.

amansinghaljpr, 2025-04-02:
hey @niyazFattahov were you able to get a fix ? I am trying to get into the luckfox world , and my goal is exactly same. But it looks like you're equally stuck . Also @luckfox-eng29 is there any github issue , or article or anything you can point me to where someone was sucessfully able to get a OV5647 camera with luckfox board ?

Olewojt, 2026-02-10:
Sadly, OV5647 RPi cameras ([Camera module 1](https://www.raspberrypi.com/documentation/accessories/camera.html#hardware-specification)) won't work out of the box with Luckfox Pico Zero.

**GPIO3_C5** that is referenced in [board schematic](https://wiki.luckfox.com/assets/files/Luckfox-Pico-Zero-2f70e728ec7d33ad9c60f72d8cff7a2b.pdf) as **MIPI/LVDS_RST** output is **1.8V** (probed by me) which is not enough to make the camera turn on apparently (you won't get I2C communication either). There are no available schematics for those camera boards as Raspberry never published them (as per this [reply](https://github.com/raspberrypi/documentation/issues/1324#issuecomment-549064549)). 

I've successfully managed to get it enabled by bridging 3.3V pin and MIPI/LVDS_RST on FPC cable directly, but be aware that it could potentially damage the board, as applying 3.3V to 1.8V pin isn't quite a good idea. Best solution would be to incorporate a MOSFET. Btw. I'm using RPi camera cable standard mini 200mm to connect it to Pico Zero.

[image]

It's get detected by _rockchip-mipi-csi2-hw_ driver but fails miserably afterwards with given stack trace:
```
[    2.242278] rk_dvbm ffa70000.rkdvbm: probe start
[    2.242318] rk_dvbm ffa70000.rkdvbm: dvbm ctx ac27f833
[    2.246969] rk_dvbm ffa70000.rkdvbm: probe success
[    2.303517] rkcifhw ffa10000.rkcif: no iommu attached, using non-iommu buffers
[    2.303552] rkcifhw ffa10000.rkcif: No reserved memory region assign to CIF
[    2.303950] rkcif rkcif-mipi-lvds: rkcif driver version: v00.02.00
[    2.304112] rkcif rkcif-mipi-lvds: attach to cif hw node
[    2.304128] rkcif rkcif-mipi-lvds: failed to get dphy hw node
[    2.304139] rkcif rkcif-mipi-lvds: rkcif wait line 0
[    2.304150] rkcif rkcif-mipi-lvds: rkcif fastboot reserve bufs num 3
[    2.304163] : terminal subdev does not exist
[    2.304175] : terminal subdev does not exist
[    2.304186] : terminal subdev does not exist
[    2.304196] : terminal subdev does not exist
[    2.304209] : get_remote_sensor: video pad[0] is null
[    2.304219] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[    2.304230] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[    2.304238] : get_remote_sensor: video pad[0] is null
[    2.304245] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[    2.304253] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[    2.304262] : get_remote_sensor: video pad[0] is null
[    2.304270] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[    2.304278] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[    2.304285] : get_remote_sensor: video pad[0] is null
[    2.304292] : rkcif_update_sensor_info: stream[0] get remote sensor_sd failed!
[    2.304301] : rkcif_scale_set_fmt: req(80, 60) src out(0, 0)
[    2.317495] rkcif rkcif-mipi-lvds: No memory-region-thunderboot specified
[    2.318762] rockchip-mipi-csi2-hw ffa20000.mipi-csi2-hw: enter mipi csi2 hw probe!
[    2.318929] rockchip-mipi-csi2-hw ffa20000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[    2.319067] rockchip-mipi-csi2-hw ffa30000.mipi-csi2-hw: enter mipi csi2 hw probe!
[    2.319184] rockchip-mipi-csi2-hw ffa30000.mipi-csi2-hw: probe success, v4l2_dev:mipi-csi2-hw!
[    2.319561] rockchip-mipi-csi2 mipi0-csi2: attach to csi2 hw node
[    2.319627] rkcif rkcif-mipi-lvds: Entity type for entity rockchip-mipi-csi2 was not initialized!
[    2.319648] rockchip-mipi-csi2: Async registered subdev
[    2.319662] rockchip-mipi-csi2: probe success, v4l2_dev:rkcif-mipi-lvds!
[    2.346177] rkisp_hw ffa00000.rkisp: is_thunderboot: 0
[    2.346205] rkisp_hw ffa00000.rkisp: Missing rockchip,grf property
[    2.346251] rkisp_hw ffa00000.rkisp: max input:0x0@0fps
[    2.346434] rkisp_hw ffa00000.rkisp: get sram size:253952
[    2.346450] rkisp_hw ffa00000.rkisp: no iommu attached, using non-iommu buffers
[    2.346461] rkisp_hw ffa00000.rkisp: No reserved memory region. default cma area!
[    2.346833] rkisp rkisp-vir0: rkisp driver version: v02.05.00
[    2.346958] rkisp rkisp-vir0: No memory-region-thunderboot specified
[    2.347968] rkisp rkisp-vir0: Entity type for entity rkisp-isp-subdev was not initialized!
[    2.384738] rockchip-csi2-dphy-hw ff3e8000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[    2.406449] rockchip-csi2-dphy csi2-dphy0: dphy0 matches ov5647 4-0036:bus type 5
[    2.408772] rkcif-mipi-lvds: rkcif_update_sensor_info: get terminal ov5647 4-0036 g_frame_interval failed!
[    2.408797] rkcif-mipi-lvds: sensor fmt invalid, set to default size
[    2.408817] 8<--- cut here ---
[    2.599025] Unable to handle kernel NULL pointer dereference at virtual address 00000000
[    2.607116] pgd = cf9a9794
[    2.609851] [00000000] *pgd=00000000
[    2.613438] Internal error: Oops - BUG: 5 [#1] THUMB2
[    2.618488] Modules linked in: phy_rockchip_csi2_dphy(+) phy_rockchip_csi2_dphy_hw video_rkisp video_rkcif ov5647 rk_dvbm
[    2.629453] CPU: 0 PID: 186 Comm: insmod Not tainted 5.10.160 #5
[    2.635457] Hardware name: Generic DT based system
[    2.640357] PC is at rkcif_output_fmt_check+0xc/0x2a4 [video_rkcif]
[    2.646671] LR is at rkcif_set_fmt+0x99/0x3d4 [video_rkcif]
[    2.652242] pc : [<af80da2c>]    lr : [<af813d69>]    psr: 60000033
[    2.658501] sp : b17839d0  ip : 00000000  fp : b02f7047
[... 58 lines cut ...]
[    3.125193] [<af813d69>] (rkcif_set_fmt [video_rkcif]) from [<af816627>] (rkcif_set_default_fmt+0xc7/0x104 [video_rkcif])
[... cut ...]
[    3.273101] [<b0043b4b>] (sys_finit_module) from [<b0008201>] (ret_fast_syscall+0x1/0x58)
[    3.281270] Exception stack(0xb1783fa8 to 0xb1783ff0)
[    3.286325] 3fa0:                   6332defc ae9adebc 00000003 005c2008 00000000 ae9adebc
[    3.294498] 3fc0: 6332defc ae9adebc ae9addc4 0000017b 00000003 00000000 005c09dc 00000000
[    3.302668] 3fe0: ae9adc28 ae9adc18 005067e8 a6eae8b0
[    3.307723] Code: f8d0 3588 7804 6845 (681a) f243
[    3.312557] ---[ end trace c3da777945d3c55b ]---
```
It seems it can't get info from sensor about g_frames_interval.