LuckfoxTECH/luckfox-pico #243: Device Tree Overlay

DansDesigns, 2025-03-07:
Hello,

I have converted the following overlay for an SSD1351 SPI OLED from my original that worked on a NanoPi Neo (Allwinner H3 Cortex-A7),
how would I add this Overlay to the Ubuntu image running on my Core1106?

with Armbian on the NanoPi Neo, the command `sudo armbian-add-overlay <overlay_name>.dts` installs a new Overlay to the system,
is the same possible with Ubuntu on the RV1106?

here is the Overlay i converted based on the wiki page: Luckfox-Pico-RV1106 / Luckfox-config, section 3. Device Tree Configuration:

```

&spi0 {
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&spi0m0_pins &spi0m0_cs0>;
	//cs-gpios = <&gpio1 RK_PC0 1>;
	// cs-gpios = <&gpio1 26 1>;
	#address-cells = <1>;
	#size-cells = <0>;
	//spidev@0 {
	//	compatible = "rockchip,spidev";
	//	spi-max-frequency = <4500000>;
	//	reg = <0>;
	//};

	ssd1351@0{
		compatible = "solomon,ssd1351"";
		reg = <0>;
		status = "okay";
		spi-max-frequency = <4500000>;
		bgr = <0>;
        	bpp = <16>;
		dc = <&gpio2 RK_PA2 1>;
		reset = <&gpio2 RK_PA3 1>;
		rotation = <0>;
		spi-cpol;
		spi-cpha;
		fps = <30>;
		buswidth = <8>;
		rgb;
		solomon,height = <128>;
        	solomon,width = <128>;
        	solomon,page-offset = <0>;
	};

};

```

DansDesigns, 2025-03-09:
UPDATE:
so I managed to get the following .dts to compile ok with the command,
 `dtc -I dts -O dtb ssd1351_spi_overlay_rv1106.dts -o ssd1351_spi_overlay_rv1106.dtb`

.dts:
``` 
/dts-v1/;
/plugin/;
&{/spi@ff500000}{
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&spi0m0_pins &spi0m0_cs0>;
	#address-cells = <1>;
	#size-cells = <0>;

	ssd1351@0{
		compatible = "solomon,ssd1351";
		reg = <0>;
		status = "okay";
		spi-max-frequency = <4500000>;
		bgr = <0>;
        	bpp = <16>;
		dc = <&gpio2_pa2 1>;
		reset = <&gpio2_pa3 1>;
		rotation = <0>;
		spi-cpol;
		spi-cpha;
		fps = <30>;
		buswidth = <8>;
		rgb;
		solomon,height = <128>;
        	solomon,width = <128>;
        	solomon,page-offset = <0>;
	};

};
```

however I am unable to write the file to `/sys/kernel/config/device-tree/overlays/spi0m0/dtbo`
with the command `cat ssd1351_spi_overlay_rv1106.dtb > /sys/kernel/config/device-tree/overlays/spi0m0/dtbo`

console returns `-bash: /sys/kernel/config/device-tree/overlays/spi0m0/dtbo: Permission denied `, even when running with sudo...

info taken from
[rv106 framebuffer](https://github.com/LitchiCheng/RV1106_Linux/blob/main/framebuffer/rv1106g-luckfox-pico-pro-max.dtsd)
 and
[luckfox forum]( https://forums.luckfox.com/viewtopic.php?p=3897&hilit=overlay#p3897)

luckfox-eng29, 2025-03-11:
您好，RV1106 的内核使用 FDT 格式打包，不支持启动 **DTC_SYMBOLS** 和 **Fixup dtb by removing empty nodes**, 所以您编写的这种动态设备树的格式是不支持传入内核的。RV1106 的动态设备树的格式不支持引用节点需要和原始的dtb比对，操作步骤较为复杂，建议还是直接在设备树中进行修改。

DansDesigns, 2025-03-12:
Hello, ah ok thank you for the detailed answer.

I have downloaded the sdk and added the contents to the existing `&spi0` section within the ultra_w.dts as follows:

``` 
/**********SPI**********/
&spi0 {
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&spi0m0_pins &spi0m0_cs0>;
	#address-cells = <1>;
	#size-cells = <0>;
	//spidev@0 {
	//	compatible = "rockchip,spidev";
	//	spi-max-frequency = <4500000>;
	//	reg = <0>;
	//};

	ssd1351@0{
		compatible = "solomon,ssd1351";
		reg = <0>;
		status = "okay";
		spi-max-frequency = <4500000>;
		bgr = <0>;
        	bpp = <16>;
		dc = <&gpio2 RK_PA2 1>;
		reset = <&gpio2 RK_PA3 1>;
		rotation = <0>;
		spi-cpol;
		spi-cpha;
		fps = <30>;
		buswidth = <8>;
		rgb;
		solomon,height = <128>;
        	solomon,width = <128>;
        	solomon,page-offset = <0>;
	};
};
```

I also have the following at the top of the file, between the wireless-bluetooth section:

``` 

/ {
	model = "Luckfox Pico Ultra W";
	compatible = "rockchip,rv1103g-38x38-ipc-v10", "rockchip,rv1106g3";

	restart-poweroff {
		compatible = "restart-poweroff";
	};

	sdio_pwrseq: sdio-pwrseq {
		compatible = "mmc-pwrseq-simple";
		pinctrl-names = "default";
		reset-gpios = <&gpio1 RK_PA2 GPIO_ACTIVE_LOW>;
	};
	
	
	/*LCD DC*/
	gpio2pa2:gpio2pa2 {
		compatible = "regulator-fixed";
		pinctrl-names = "default";
		pinctrl-0 = <&gpio2_pa2>;
		regulator-name = "gpio2_pa2";
		regulator-always-on;
	};

	/*LCD RESET*/
	gpio2pa3:gpio2pa3 {
		compatible = "regulator-fixed";
		pinctrl-names = "default";
		pinctrl-0 = <&gpio2_pa3>;
		regulator-name = "gpio2_pa3";
		regulator-always-on;
	};

  wireless_bluetooth: wireless-bluetooth {
    compatible = "bluetooth-platdata";
    uart_rts_gpios = <&gpio0 RK_PA5 GPIO_ACTIVE_LOW>;
    pinctrl-names = "default", "rts_gpio";
    pinctrl-0 = <&uart1m0_rtsn>;
    pinctrl-1 = <&uart1_gpios>;
    BT,wake_gpio     = <&gpio0 RK_PA2 GPIO_ACTIVE_HIGH>;
    BT,wake_host_irq = <&gpio1 RK_PA2 GPIO_ACTIVE_HIGH>;
    status = "okay";
  };
};

&pinctrl{
	/*LCD DC*/
	gpio2-pa2 {
		gpio2_pa2:gpio2-pa2 {
			rockchip,pins = <2 RK_PA2 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

	/*LCD RESET*/
	gpio2-pa3 {
		gpio2_pa3:gpio2-pa3 {
			rockchip,pins =	<2 RK_PA3 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};
};
```

will these work or do i need to modify some other bits too??

thank you for your help

DansDesigns, 2025-03-13:
ah excellent, I have been using luckfox-config to turn the RGB off but those entries for the dts will make it alot easier, thank you for that bit of coding gold - I will add those to my .dts now..

I'm sorry, I don't fully understand what you mean by, `the .dts can be adjusted according to the printing information of the kernel to determine whether the driver is loaded successfully`

- what is the printing information you are refering to?
- will i need to use something like `modprobe fb_ssd1351` to activate the display?

luckfox-eng29, 2025-03-14:
1 Printing information refers to the information that will be printed to the terminal when the Luckfox Pico loads the kernel, which can be viewed again by entering the terminal through the "dmesg" command or "cat /var/log/messages", if the driver of "solomon, ssd1351" fails to load, there is generally a print-related error message
2 If you use y in the kernel's menuconfig to compile the driver directly to the kernel, you don't need to execute modprobe fb_ssd1351'' after entering the system; If you use m to compile and generate a .ko file, you need to use insmod to load the driver in the /oem/usr/ko/ directory on the board side, and the system provided by default does not support modprobe to load the driver

In addition, **luckfox-config** is only for the original DTS file, if the DTS file is modified, try not to use **luckfox-config** to prevent **luckfox-config** from conflicting with the modified DTS file when used

DansDesigns, 2025-03-14:
ah I understand now, thank you.

I have enabled the SSD1351 section within `drivers/staging/fbtft` in kernelconfig (I noticed there are 2 listings for `FB driver for the SSD1351 LCD Controller`), I wasnt sure which one to enable so i enabled both?.. I have also disabled RGB and Camera in kernel, this solved the issue of it needing to be disabled in luckfox-config.

running `dmesg | grep ssd1351` did not show any reference to ssd1351, i also tried with fb_tft_ssd1351 but nothing..

would a copy of the dmesg be helpful?
also cat `/var/log/messages` states the location does not exist?

luckfox-eng29, 2025-03-15:
No /dev/fb* indicates that the driver of FBTFT has not been loaded successfully, first determine whether the SPI is successfully registered and then add the printing code from the driver source code to determine where the loading failed, more details need the hardware environment to determine that I can't help.

DansDesigns, 2025-03-15:
Thank you for your help thus far, it is greatly appreciated!

```
dmesg` | grep spi
[   21.634365] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/status
[   21.634402] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/pinctrl-0
[   21.634419] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/spidev@0/status
[   21.634434] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/spidev@0/spi-max-frequency
[  721.729090] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/status
[  721.729129] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/pinctrl-0
[  721.729147] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/spidev@0/status
[  721.729162] OF: overlay: WARNING: memory leak will occur if overlay removed, property: /spi@ff500000/spidev@0/spi-max-frequency
```

it seems the max-frequency is not accepted - I have re-checked the data sheet for the ssd1351 and it requires 20000000 or 20MHz, this is much higher than my stated 4500000, so not too sure why it is not accepted or why the same error happens twice..

i have updated the .dts file to reflect the change in frequency for the next compilation,
however upon `cat /usr/bin/luckfox-config` I also see that:

```
function` luckfox_FBTFT() {
	local fbtft_enable
	fbtft_enable=$(dialog --menu "FBTFT Enable Config" 10 40 2 \
		0 "disable" \
		1 "enable" \
		2>&1 >/dev/tty)
	if [ -z "$fbtft_enable" ]; then
		luckfox_advanced_options
	fi

	if [ "$fbtft_enable" == 1 ]; then
		fbtft_mode=$(dialog --menu "FBTFT MODE Config" 10 40 2 \
			0 "st7789v/st7789vm" \
			2>&1 >/dev/tty)
		if [ -z "$fbtft_mode" ]; then
			luckfox_advanced_options
		elif [ "$fbtft_mode" == 0 ]; then
			fbtft_compatible="sitronix,st7789v"
		fi
	fi
```

it looks like the kernel is still looking to the st7789v even though i have disabled that specific selection within 
`sudo ./build.sh kernelconfig`

I am going to re-download the SDK and start again..
are you able to confirm the syntax and layout of the ssd1351 overlay i have? 

```
/**********SPI**********/`

&rgb {
	status = "disabled";
};

&panel {
	status = "disabled";
};

&spi0 {
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&spi0m0_pins &spi0m0_cs0>;
	#address-cells = <1>;
	#size-cells = <0>;
	
	//spidev@0 {
	//	compatible = "rockchip,spidev";
	//	spi-max-frequency = <4500000>;
	//	reg = <0>;
	//};

	ssd1351@0{
		compatible = "solomon,ssd1351";
		reg = <0>;
		status = "okay";
		spi-max-frequency = <20000000>;
		bgr = <0>;
        	bpp = <16>;
		dc = <&gpio2 RK_PA2 1>;
		reset = <&gpio2 RK_PA3 1>;
		rotation = <0>;
		spi-cpol;
		spi-cpha;
		fps = <30>;
		buswidth = <8>;
		rgb;
		solomon,height = <128>;
        	solomon,width = <128>;
        	solomon,page-offset = <0>;
	};
};
```

DansDesigns, 2025-03-15:
I have recompiled a fresh install of the sdk with ssd1351 enabled in kernelconfig, i have edited both `dts-config`: 

```
&spi0 {
	status = "okay";
	pinctrl-names = "default";
	pinctrl-0 = <&spi0m0_pins &spi0m0_cs0 &spi0m0_cs1>;
	#address-cells = <1>;
	#size-cells = <0>;
	
	//spidev@0 {
	//	status = "okay";
        //	compatible = "rockchip,spidev";
	//	spi-max-frequency = <20000000>;
	//	reg = <0>;
	//};
	
	fbtft@0 {
		status = "okay";
	};
    
	ssd1351@0{
		compatible = "solomon,ssd1351";
		reg = <0>;
		status = "okay";
		spi-max-frequency = <20000000>;
		bgr = <0>;
        	bpp = <16>;
		dc-gpios = <&gpio2 RK_PA2 1>;
		reset-gpios = <&gpio2 RK_PA3 0>;
		rotation = <0>;
		spi-cpol;
		spi-cpha;
		fps = <30>;
		buswidth = <8>;
		rgb;
		solomon,height = <128>;
        	solomon,width = <128>;
        	solomon,page-offset = <0>;
	};
};
&pinctrl{
	/*LCD DC*/
	gpio2-pa2 {
		gpio2_pa2:gpio2-pa2 {
			rockchip,pins = <2 RK_PA2 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};

	/*LCD RESET*/
	gpio2-pa3 {
		gpio2_pa3:gpio2-pa3 {
			rockchip,pins =	<2 RK_PA3 RK_FUNC_GPIO &pcfg_pull_none>;
		};
	};
};
```

and `rv1106-luxkfox-pico-ultra-ipc.dtsi`: 

```

/*****************************PINCTRL********************************/

// SPI
&spi0 {
	pinctrl-0 = <&spi0m0_clk &spi0m0_miso &spi0m0_mosi &spi0m0_cs0>;
	#address-cells = <1>;
	#size-cells = <0>;
	spidev@0 {
		compatible = "rockchip,spidev";
		spi-max-frequency = <50000000>;
		reg = <0>;
	};

  fbtft@0{
    compatible = "solomon,ssd1351";
    reg = <0>;
    spi-max-frequency = <20000000>;
    debug = <0x7>;
    bgr = <0>;
    bpp = <16>;
    dc-gpios = <&gpio2 RK_PA2 1>;
    reset-gpios = <&gpio2 RK_PA3 0>;
    rotation = <0>;
    spi-cpol;
    spi-cpha;
    fps = <30>;
    buswidth = <8>;
    rgb;
    solomon,height = <128>;
    solomon,width = <128>;
    solomon,page-offset = <0>;
    
    // OG pins
    //led-gpios = <&gpio2 RK_PB0 GPIO_ACTIVE_HIGH>;//BL
    //dc-gpios = <&gpio2 RK_PB1 GPIO_ACTIVE_HIGH>;      //DC
    //reset-gpios = <&gpio1 RK_PC3 GPIO_ACTIVE_LOW>;    //RES
  };
};

```

but still when `cat /usr/bin/luckfox-config`:

```
function luckfox_FBTFT() {
	local fbtft_enable
	fbtft_enable=$(dialog --menu "FBTFT Enable Config" 10 40 2 \
		0 "disable" \
		1 "enable" \
		2>&1 >/dev/tty)
	if [ -z "$fbtft_enable" ]; then
		luckfox_advanced_options
	fi

	if [ "$fbtft_enable" == 1 ]; then
		fbtft_mode=$(dialog --menu "FBTFT MODE Config" 10 40 2 \
			0 "st7789v/st7789vm" \
			2>&1 >/dev/tty)
		if [ -z "$fbtft_mode" ]; then
			luckfox_advanced_options
		elif [ "$fbtft_mode" == 0 ]; then
			fbtft_compatible="sitronix,st7789v"
		fi
	fi
```

and no `/dev/fb0`

DansDesigns, 2025-03-16:
hello again,
so it appears the file `luckfox-pico/project/cfg/BoardConfig_IPC/overlay/overlay-luckfox-config/usr/bin/luckfox-config` also needs lines 367 & 375 changing from `"sitronix,st7735r"` to `"solomon,ssd1351"` as this doesnt get automatically updated when choosing different framebuffer hardware within kernelconfig.

so far the files within the SDK that need editing are:

rv1106-luckfox-pico-ultra-ipc.dtsi
dts_config (after running `sudo ./build.sh lunch)`
BoardConfig-EMMC-Ubuntu-RV1106_Luckfox_Pico_Ultra_W-IPC.mk
luckfox-config (SDK version mentioned above)

`fbcon`, `ssd1351 fb device` and `support for small framebuffer devices` are enabled in kernelconfig,
`dmesg | grep fb`, `dmesg | grep spi` & `dmesg | grep ssd1351` shows nothing
and when running `lsmod` there is no reference to framebuffer, fbcon, fbtft, solomon or ssd1351
and no `/dev/fb0`

am i missing something in kernelconfig or another file to enable fb?

luckfox-eng29, 2025-03-17:
I can't provide an explanation for all of your development problems, your development goals require you to have some basic knowledge of driver porting.

My suggestion now is to ditch the use of luckfox-config altogether, as it will lead to all sorts of things you can't tell, edit the ```.BoradConfig.mk``` before the SDK compiles the image and remove the ```overlay-luckfox-config``` from the ```RK_POST_OVERLAY```.

Secondly, you have modified the content of the spi0 node in DTSI and DTS, and it is difficult to judge the final state when it is finally compiled into DTB, so it is recommended to delete the entire SPI0 node in DTSI and configure it in DTS. The last generated dtb file is <SDK>in /sysdrv/source/objs_kernel/arch/arm/boot/dts/, you can run ```fdtdump XXX.dtb | less``` to see if the final configuration meets your expectations.

dmesg only sees the latest kernel activity information, the information in the SPI initialization part may  be don't displayed, the information obtained by ```cat /var/log/messages``` is complete, I have measured that the Buildroot system can use this command, Maybe the Ubuntu system can have other ways to view. The most reliable solution is to use the USB to UART tool to boot the system from the serial terminal of the Core1106, which will print the complete kernel information.

lsmod can only view the drivers installed with ```insmod *.ko```, and the drivers you load are compiled directly into the kernel

I can't be sure if your device tree syntax is correct, because the device tree files that can be used correctly need to be debugged step by step by combining kernel logs and driver source code

DansDesigns, 2025-03-17:
Thank you for the detailed reply, while i do have basic knowledge in dtoverlay writing i have never worked with a system that didnt allow dynamic dts loading so this whole thing is admittedly a new area to me. 

For my current project i am going to switch back to the NanoPi Neo as that system works the way i need.
I will be looking to upgrade my other project to the luckfox core1106 in the future - that doesnt require a screen so hopefully will be alot easier!

Thank you again

closed 2025-03-17 by DansDesigns