LuckfoxTECH/luckfox-pico #245: rk3506 build issue on WSL

bousong, 2025-03-10:
I have encountered a build problem with Ubuntu-22.04 running on WSL2. I have not seen RK3506 SDK on github yet, so I decided to post it in here instead.

############### Rockchip Linux SDK ###############

Manifest: luckfox_linux6.1_rk3506_release_v1.0_20241212.xml

Log colors: message notice warning error fatal

WSL is not supported!

ERROR: Running ./build.sh - check_sdk failed!
ERROR: exit code 1 from line 417:
    "$RK_SCRIPTS_DIR/check-sdk.sh"
ERROR: call stack:
    build.sh: check_sdk(417)
    build.sh: main(533)
    build.sh: main(793)

luckfox-eng33, 2025-03-11:
You can directly cancel this judgment and compile normally. We will make adjustments later so that the SDK can be compiled normally in the WSL2 environment.
This code comes from RK. Maybe RK doesn't want to use WSL to reduce their workload.

bousong, 2025-03-11:
I see. Thanks for your response.

closed 2025-03-11 by bousong