LuckfoxTECH/luckfox-pico #248: Low FPS with Pico Ultra W

conner-replogle, 2025-03-13:
I am attempting to use the https://github.com/LuckfoxTECH/luckfox_pico_rkmpi_example
And am barely getting 10fps, I am targeting 30 is there anything I can do to improve it?

luckfox-eng33, 2025-03-15:
I think you can see this [wiki](https://wiki.luckfox.com/Luckfox-Pico/RKMPI-example/)

conner-replogle, 2025-03-16:
I ran that demo specifically the rtsp opencv one and I was getting 10 fps, when I use the rkipc that comes default I get 30 fps.

luckfox-eng29, 2025-03-17:
Hello， rkipc only uses the components of rkmpi to complete, which is less flexible but basically does not copy in the middle, https://github.com/LuckfoxTECH/luckfox_pico_rkmpi_example In order to introduce the use of opencv-mobile to simplify the process of image processing, it is indispensable to ```read the image``` > ```process``` - > ```copy back to the buffer```, if you want to pursue efficiency, please refer to it Rockchip's official documentation

conner-replogle, 2025-04-29:
For people who see this later the other example https://github.com/LuckfoxTECH/luckfox_pico_rkmpi_example/blob/kernel-5.10.160/example/luckfox_pico_rtsp_opencv_capture/ Works great if you remove opencv. What I learned is this board can handle 30fps easily as long as your video pipeline is very streamlined, I  write frames to a Unix socket and then read it in from another process and can handle 30fps with no latency.