LuckfoxTECH/luckfox-pico #249: Problem with Luckfox Pico Plus Not Detecting SD Card in Soctoolkit

boozary, 2025-03-13:
I have a Luckfox Pico Plus board. When I insert an SD card, it does not show up in the SD memory section of Soctoolkit, so I can't install the image on it.

However, when I install the OS on NAND, Linux itself mounts the SD card, and I can use it normally.

Is there anything specific I need to do to be able to burn the image to the SD card?

Thanks!

luckfox-eng33, 2025-03-15:
Did you use an SD card reader? I think the [WIKI](https://wiki.luckfox.com/Luckfox-Pico/Luckfox-Pico-RV1103/Luckfox-Pico-Plus-Mini/Luckfox-Pico-SD-Card-burn-image) has already explained it clearly.

boozary, 2025-03-15:
Hello,
Thanks for your response. I thought the SD card would show up if it was inserted into the Luckfox itself. But after your suggestion, I tried putting the SD card in a card reader, and it was recognized.

I really appreciate it.