LuckfoxTECH/luckfox-pico #254: Ubuntu Camera Error

B3B3K, 2025-03-25:
I have changed env.img variables. (CMA size to 10mb) but i cant get camera to work. I can see camera on /dev folder as video0 etc. but i cant acces them. Here is the dmesg some dmesg:

`
[   25.747137] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[3] get remote terminal sensor failed!
[   25.756289] stream_cif_mipi_id3: update sensor info failed -19
`

`
[   19.557170] 16384 pages RAM
[   19.557173] 0 pages HighMem/MovableOnly
[   19.557178] 4857 pages reserved
[   19.557181] 3072 pages cma reserved
[   19.557222] rkisp: probe of rkisp-vir0 failed with error -12
[   20.026471] rockchip-csi2-dphy-hw ff3e8000.csi2-dphy-hw: csi2 dphy hw probe successfully!
[   20.158671] rockchip-csi2-dphy csi2-dphy0: csi2 dphy0 probe successfully!
[   22.216221] rkcif rkcif-mipi-lvds: clear unready subdev num: 3
[   22.219762] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[   22.219793] rkcif-mipi-lvds: Async subdev notifier completed
[   22.219817] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[   22.219826] rkcif-mipi-lvds: There is not terminal subdev, not synchronized with ISP
[   22.219956] rkcif-mipi-lvds: rkcif_update_sensor_info: stream[0] get remote terminal sensor failed!
[   22.219968] rkcif-mipi-lvds: There is not terminal subdev, not synchronized with ISP
`

luckfox-eng33, 2025-03-25:
CMA is too small, the recommended size is 36MB

B3B3K, 2025-03-26:
> CMA is too small, the recommended size is 36MB

well, changed CMA to 32 MB via Uboot. So on start, it cant get pins state correctly for camera. 

[   52.685556] sc3336 4-0030: could not get default pinstate
[   52.685584] sc3336 4-0030: could not get sleep pinstate
[   56.273104] rkcif_sditf rkcif-mipi-lvds-sditf: failed attach cif dev

B3B3K, 2025-03-26:
Changed to 36M (CMA) but same error.

B3B3K, 2025-03-31:
OH, forgot enable CSI on luckfox-config lol, my mistake sorry

closed 2025-03-31 by B3B3K