LuckfoxTECH/luckfox-pico #257: Bluetooth Advertising Issue on Luckfox Lyra Ultra W (Ubuntu 22.04)

hongthaiovi, 2025-03-30:
Issue Description

I am experiencing an issue with Bluetooth advertising on the Luckfox Lyra Ultra W running Ubuntu 22.04. While the board can scan and detect other devices, it does not appear on other devices when advertising.

System Information

OS: Ubuntu 22.04

Bluetooth Controller: AIC8800DC

Kernel Version: 6.1.99

Driver in Use: btusb

Logs and Observations

Below are the relevant logs from dmesg | grep -i bluetooth:

```
[    2.702051] Bluetooth: RFCOMM TTY layer initialized
[    2.702123] Bluetooth: RFCOMM socket layer initialized
[    2.702180] Bluetooth: RFCOMM ver 1.11
[    2.702220] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[    2.702253] Bluetooth: BNEP socket layer initialized
[    2.702689] [BT_RFKILL]: bluetooth_platdata_parse_dt: uart_rts_gpios is no-in-use.
[    2.702756] [BT_RFKILL]: bluetooth_platdata_parse_dt: get property: BT,power_gpio = 53.
[    2.702808] [BT_RFKILL]: bluetooth_platdata_parse_dt: clk_get failed!!!.
[    2.703259] input: bt-powerkey as /devices/platform/wireless-bluetooth/input/input1
[   16.231262] Bluetooth: MGMT ver 1.22
[   20.142359] Bluetooth: MGMT ver 1.22
[   22.855789] aic_bluetooth_mod_init

```
Configuration

My device tree configuration:

```
wireless_bluetooth: wireless-bluetooth {
    compatible = "bluetooth-platdata";
    BT,power_gpio = <&gpio1 RK_PC5 GPIO_ACTIVE_HIGH>;
    status = "okay";
};
```

Issue Analysis

clk_get failed!!! error: This suggests a missing or misconfigured clock reference in the device tree.

Advertising not visible: Even though hciconfig and bluetoothctl show that advertising is enabled, the board does not appear on other devices.

Scanning works: The device is able to scan and detect other Bluetooth devices.

Steps Taken

Checked btusb driver using modinfo btusb.

Verified Bluetooth service status with systemctl status bluetooth.

Tried setting the device as discoverable:
```

bluetoothctl
power on
discoverable on
advertising on
```

Ensured that no firewall or power management settings were blocking Bluetooth.

Request for Help

Has anyone encountered a similar issue with the AIC8800DC chipset or the clk_get failed!!! error? Any guidance on resolving this would be greatly appreciated!

luckfox-eng29, 2025-04-07:
Hello luckfox-lyra-ultra-w "bluetoothctl" under the ubuntu image does not support the "advertising on" directive, you need to use "menu advertising" to enter the submenu for configuration.
We have not actually tested BLE broadcasting, so there is no relevant operation guidance to provide for the time being, please refer to [bluetoothctl-advertise](https://www.mankier.com/1/bluetoothctl-advertise) for detailed operation.
It is recommended that please report problems related to Luckfox Lyra under [luckfox forums](https://forums.luckfox.com/viewforum.php?f=17), which is more convenient to follow up the progress of problem solving, and is mainly used here to report luckfox-pico problems

hongthaiovi, 2025-04-07:
Thanks @luckfox-eng29 , let me check