LuckfoxTECH/luckfox-pico #258: Luckfox-Pico-Ultra-W wifi AP speed 40Mbps？

bltzmnn, 2025-03-31:
Only STA mode 100Mbps? Wifi AP mode,speed only about 40Mbps,is it OK?

luckfox-eng29, 2025-04-07:
Hello, we have measured the speed of the Luckfox-Pico-Ultra-W in STA mode up to 50Mbps in ideal conditions, and in AP mode, although we have not measured it, it will not exceed 30Mbps in theory

bltzmnn, 2025-04-07:
???
The [wiki](https://wiki.luckfox.com/zh/Luckfox-Pico/Luckfox-Pico-Ultra-W-WIFI) says:It is about 100Mbps in STA mode! And mine,in STA mode also about 100Mbps!

bltzmnn, 2025-04-07:
Please do a test.

luckfox-eng29, 2025-04-07:
The wiki only provides the usage of iperf3, the test results are the ethernet port, our network environment test to increase the SDIO rate to 100MHz can not be stable at 60Mbps, please check whether the IP you use is wlan0

bltzmnn, 2025-04-07:
Yes,it is eth0.So in AP mode ,speed 40Mbps is OK? And please modified the wiki.

luckfox-eng29, 2025-04-07:
In AP mode, the actual test is only about 20Mbps, and the actual performance is affected by the network environment

bltzmnn, 2025-04-07:
OK，mine AP mode speed(iperf3,hostapd 2.11) is about 40Mbps. Maybe you need a newer version of hostapd.

zagfai, 2025-04-24:
That is not Wifi6-.-