LuckfoxTECH/luckfox-pico #263: RV1103 Audio Video RTSP issues

pauleffect90, 2025-04-15:
Hello.

Since the Luckfox Pico Plus (RV1103 based) has the audio codec pins mapped to testpoints, I added a single ended microphone. I followed the configs done on the Ultra W.

I can succesfully record audio via rk_mpi_ai_test. [Audio sample](https://jumpshare.com/s/2kAXcXFFpKJn3uzF61DB)
`rk_mpi_ai_test --sound_card_name=hw:0,0 --device_rate=16000 --device_ch=2 --out_rate=16000 --out_ch=2 --output=/tmp`

Now, to test the audio + video workflow, i modified the rkipc.ini file in /userdata

`[audio.0]
enable = 1
card_name = hw:0,0
encode_type = G711A
format = S16
# sample_rate = 8000
sample_rate = 16000
channels = 1
frame_size = 1152
bit_rate = 32000
input = mic_in
volume = 50
enable_aed = 0
enable_bcd = 0
enable_vqe = 0
vqe_cfg = /oem/usr/share/vqefiles/config_aivqe.json`

I tried different sample rates/ encoders (G711A, MP2, MP3), frame sizes etc. and found the only one to output _something_ is **G711A**. I tried with enable_vqe both enabled and disabled. 

The sound quality is ok, but there's very strong, constant horizontal bands at fixed high freq. on the spectrogram - simply put, [beeps](https://jumpshare.com/s/jH4tjvfJ1vKxBnn7q0Mf). Lots and lots and lots of beeps I have no clue how to juggle around of. Using VQE and specifically anr (auto noise reduction?) eliminates this, along with 90% of the actual audio (sadly).

I'm sure there must be something trivial I'm completely missing.

Thank you for your time.

luckfox-eng29, 2025-04-29:
Hello, we currently lack sufficient debugging experience in audio and have not tested audio 3A algorithms or noise reduction algorithms.
From a general perspective, if the raw PCM audio data obtained via rk_mpi_ai_test does not exhibit this issue, it is likely related to encoding or streaming. However, if the issue is already present in the raw PCM data, it may be caused by hardware or ALSA mixer settings.

I tested using the onboard MEMS microphone on the Luckfox Pico Ultra, and it was able to transmit audio within the normal frequency range correctly. However, I’m not sure whether there are any issues specifically in the high-frequency range.

pauleffect90, 2025-04-29:
Thank you for your time investigating this.

Let me clarify further - recording audio alone via rk_mpi_ai_test and ffmpeg works normally, clean sound. But only if video streaming is not running (by running RkLunch-stop.sh).

Audio recording gets broken when running in parallel with video - it does not matter if audio is included in the rtsp server or separately (via ffmpeg, for example).

The audio is corrupted both when streaming or when recorded to internal storage/ sd card. So it's definitely not a streaming issue or a 3A issue. I have also tested a differential microphone, change the codec settings accordingly, the issue persists.

If video + audio works on the pico ultra, the only possible explanation (that makes sense to me) is ... lack of resources? Specifically ram? The nature of the noise screams "digital".

Could you please confirm that when you recorded audio the video was also running? Because, as I said, this issue does not manifest when not running the video application.

luckfox-eng29, 2025-04-29:
I enabled audio by modifying rkipc.ini and used VLC to pull the stream and check the audio, confirming that video streaming is also working.

pauleffect90, 2025-04-29:
Thank you kindly for confirming this - my ultra w is scheduled to arrive one of these days, so I had no way of testing on RV1106.

closed 2025-04-29 by pauleffect90