LuckfoxTECH/luckfox-pico #265: USB-A Port did not work on non-standard APC USB Cable

jomud9, 2025-04-20:
Hi there. Just got a Luckfox Pico Ultra (with PoE module) and managed to flash Ubuntu image so I could do something with pkg support.

In short I would like to use this little device to pair with my APCUPS, so the custom USB Cable could connect to Luckfox USB-A Port and served an apsupsd server.

However seems that cable can not be detected. 

Tried the following
- Disable USB Networking RNDIS 
- toggle host / device mode via luckfox-config

I know the USB-A/C port stuffs so I tried both but did not work also.

Do any suggestion for these case?

謝謝各位.

luckfox-eng29, 2025-04-29:
Hello, although we haven’t used APC USB devices ourselves, we can confirm that the Luckfox Pico must be set to USB Host mode.
You should be able to see the connected device using the lsusb command.
With that as a prerequisite, you can then check whether the appropriate drivers are enabled. Typically, USB_HID support is required in the kernel.

jomud9, 2025-04-29:
Thanks for your update.
I am now using Ubuntu image. Let me figure out how to enable USB_HID support ar kernel on coming 51 Holiday. 😊

Will keep you updated