LuckfoxTECH/luckfox-pico #268: luckfox Pico Mini A USB wifi adapter

Antom91, 2025-04-24:
Is it possible? How do I add a USB to the Luckfox Pico Mini A? Any instructions? i can't find 5v on board

luckfox-eng29, 2025-04-29:
Hello, 
For the Luckfox Pico Mini A with an additional USB port, expansion can only be achieved by connecting a USB hub through the Type-C interface. Power must be supplied to both the Luckfox Pico Mini and the USB hub via the VBUS pin.
Please note that the RV1103 chip used in the Luckfox Pico Mini provides only a single native USB interface. Therefore, the USB mode must be configured to HOST mode to ensure proper operation of the USB hub.
Once in HOST mode, ADB and RNDIS functionalities will no longer be available; all control and debugging operations must be performed via the serial debug port.

Antom91, 2025-04-29:
Got it. So, is there no way to add an external USB in GPIO? Only USB from Type-C ? correct?

luckfox-eng29, 2025-04-29:
The RV1103 has only one USB interface, which is already used for the Type-C port and is not connected to any pin headers, so USB expansion can only be done through this port.

Antom91, 2025-04-29:
got it, thank you for explonation, it is very helpful information for me

closed 2025-04-29 by Antom91