LuckfoxTECH/luckfox-pico #269: Luckfox Pico Pi Like design SD Card Not working

conner-replogle, 2025-04-25:
I've created a custom PCB for the Luckfox Pico Pi that incorporates an SD card slot, mirroring the reference design and im using the Core1106. After flashing the standard Pico Pi firmware, the system boots successfully, but the SD card isn't recognized. The boot log indicates a pin conflict, specifically that I2C4 is already using the pin required for SDIO initialization, as shown by the error messages:

```
[    0.519152] rockchip-pinctrl pinctrl: pin gpio2-1 already requested by ff470000.i2c; cannot claim for ff9a0000.mmc
...
[    0.519194] rockchip-pinctrl pinctrl: could not request pin 65 (gpio2-1) from group sdmmc1m0-bus4  on device rockchip-pinctrl
```

While my schematic closely follows the official Pico Pi design (with some component removals and layout adjustments), I'm unsure if this issue stems from a firmware incompatibility, a known problem with the standard Luckfox Pico Pi firmware when using the SD card slot, or a potential error in my custom PCB schematic. Could you provide some insight?

pauleffect90, 2025-04-27:
Don't you think the error log is kinda self-explaining? Search for that node in the dts files and disable it.

conner-replogle, 2025-04-27:
It is and I did that and it does not work however why is it enabled as default for the Pico Pi. You obviously didn't read my question I'm asking if this is expected behavior for the pico pi firmware to default to having i2c block the sdcard on launch

luckfox-eng29, 2025-04-29:
Hello, you can either set the status of i2c4 to "disabled" or properly configure the pinctrl-0 of i2c4 to allow the SD card slot to function correctly. 
By default, in order to enable dynamic device tree functionality with luckfox-config, all available pin multiplexing options are included in i2c4’s pinctrl-0. Since i2c initializes earlier than sdmmc, this causes the sdmmc initialization to fail.
Originally, this design required users to determine the CSI (Camera Serial Interface) status using luckfox-config after the first system boot, and only after confirming the CSI status could other functions work as expected. However, we have now found a better solution to replace this design, and we will update the SDK accordingly in the near future.

conner-replogle, 2025-04-29:
Thank you for the response. I solved my issue.

closed 2025-04-29 by conner-replogle