LuckfoxTECH/luckfox-pico #273: Ubuntu download link not present in google drive

bhupiister, 2025-05-07:
Please share the download link for LuckFox Pico Ultra W.
Whole ubuntu folder is missing from google drive.

luckfox-eng29, 2025-05-12:
Due to licensing constraints, we can no longer provide Ubuntu support for Luckfox Pico. Thank you for your understanding and continued support.

bhupiister, 2025-05-12:
Licensing issue with ubuntu?
How about any debian image?
Using buildroot and adding custom packages is too much of work, especially in prototyping phase.

luckfox-eng29, 2025-05-12:
We understand that adding third-party libraries in Buildroot can be a cumbersome process. .While we are actively exploring alternatives to Ubuntu, we cannot guarantee that a suitable replacement will be found. However, any progress will be shared on our Wiki and the repository's main page.

vjoco, 2025-05-20:
I was porting Alpine to LuckFox Pico Ultra W. Not easy but doable... My question: can we still build the Ubuntu image?

vjoco, 2025-05-22:
so, is Ubuntu still buildable in the sdk?

luckfox-eng29, 2025-05-22:
We no longer provide support for the construction of Ubuntu on luckfox pico

vjoco, 2025-05-23:
I understand. Let's port Alpine then for all luckfox pico systems, not only rv1103 :)

bhupiister, 2025-05-23:
I was thinking of yocto support as product designers/builders incline towards support for yocto rather than buildroot. Adding package is also easy via devtool. 

I like the idea of Alpine too for the less techie developers who just want to have package manager.

papercodeIN, 2025-06-26:
Luckfox Lyra: https://drive.google.com/drive/folders/1Ol8PbmypbkcG8ZZYeNS5nyGcESkd8C8v
Luckfox Pico: https://drive.google.com/drive/folders/14kFWY93MZ4Zga4ke2PVQgUs1y9xcMG0S

AliMoal, 2025-07-10:
> > Now without ubuntu image is too pain end Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,7)
> 
> buildroot is not bad, try it. If you really really want Ubuntu, it can be ported, but with way more pain

Hi, could you tell me what you did anything about poring Alpine or yocto? I cannot go with Buildroot anymore.