LuckfoxTECH/luckfox-pico #278: SDK bound to x86 host?

plan44, 2025-05-17:
I tried to build the image for the Luckfox 86 Box/Panel on a M2 (aarch64) host system, and found that this is impossible because the `arm-rockchip830-linux-uclibcgnueabihf` toolchain comes precompiled for x86 hosts only, and also has x86 specific dependencies in Ubuntu (`gcc-multilib`, `g++-multilib`) that do not exist in aarch64 ubuntu.

Is there a way to get/build a aarch64 based toolchain and use that with the SDK?

luckfox-eng29, 2025-05-19:
Hello, since arm-rockchip830-linux-uclibcgnueabihf is a non-open cross-compilation tool provided by Rockchip, currently, if you want to develop Luckfox Pico on the MAC M series machines, you can only consider using a virtual machine platform. Since we have no experience using it on MAC, we regretfully cannot provide you with recommendations for the virtual machine platform.

dsseng, 2025-05-19:
> Hello, since arm-rockchip830-linux-uclibcgnueabihf is a non-open cross-compilation tool provided by Rockchip, currently, if you want to develop Luckfox Pico on the MAC M series machines, you can [...]

Hello! Could you please request this tool sources from Rockchip? It says it is GCC 8.3.0, so it's GPLv3+ and you have the right for sources, you can request them in writing for any derivative work of the GPL-licensed product you receive.

My use case to have some ability to configure the toolchain is using Glibc or Musl to use NetworkManager and other programs not supported on uClibc.

Thank you!

luckfox-eng29, 2025-05-20:
Hello, if you need the relevant source code, you can consult Rockchip for it. The materials we have obtained only include the SDK and the publicly available documentation on the Wiki. However, the information obtained through the -v command can already basically restore the version information of this cross-compilation toolchain.
The arm-rockchip830-linux-uclibcgnueabihf- cross-compilation tool is not a mandatory compilation tool for RV1103/RV1106. Root file systems built with other versions of uclibc glibc or musl tools (targeting armv7) can also run on RV1103/RV1106. However, the key calls rkisp and the hardware-coded rockit library are all built based on arm-rockchip830-linux-uclibcgnueabihf-, and the rockit library is explicitly not open source. Therefore, our SDK only supports building images using arm-rockchip830-linux-uclibcgnueabihf- to support all functions.
We have no experience in cross-compilation tools and do not have sufficient ability and energy to provide support for other cross-compilation tools or re-production.

plan44, 2025-05-20:
Thank you for the valuable information about the rockchip ecosystem!

I would very much like to get in contact with Rockchip to ask for the source code of the toolchain, which must be, as @dsseng said, must be opensource.

I understand that there are proprietary parts that cannot be part of the opensource, but it would help a lot if these were cleanly separated from the gnu toolchain. I am evaluating Luckfox now for 9 months, and while the hardware is excellent, the dependencies on x86 host hardware, proprietary flashing tools and a lot of binary stuff all over the SDK makes it a very difficult sell for actually using in our products.

Regarding the toolchain: I found @joshbowyer's [github repo](https://github.com/joshbowyer/arm-rockchip830-linux-uclibcgnueabihf/tree/main/share) providing a [crosstool-ng](https://github.com/crosstool-ng/crosstool-ng) based `arm-rockchip830-linux-uclibcgnueabihf` toolchain. Josh sais in the readme that the original rockchip toolchain is also based on crosstool-ng, however I don't think rockchip provides the crosstool-ng config for that.

And while I could rebuild Josh's toolchain for aarch64 (Apple Silicon Mx), the output lacks things the luckfox SDK apparently needs, like a .zip file with some libraries.

@luckfox-eng29 I'm open to help in open source efforts to improve the SDK situation around Luckfox hardware. In particular, the [OpenWrt for Luckfox](https://github.com/LuckfoxTECH/luckfox-pico) looks promising and I am right now trying to make it work on LuckFox 86 Panel, as I'd like to port my home automation platform [p44-xx-open](https://github.com/plan44/p44-xx-open).

dsseng, 2025-05-20:
Iirc Rockchip has an open tool for flashing to eMMC via boot ROM, if you are asking about that. Take a look at Rockchip GitHub.

I would also like to experiment with Yocto + meta-rockchip when I get spare time after implementing the device using the Buildroot SDK, mainly because I would prefer Yocto for some of the projects I consider Core1106 for.

@luckfox-eng29 , could you please clarify what the kernel in the SDK is based on? It doesn't seem to be linux-rockchip but rather some Android GKI 5.10. Was it provided by Rockchip? 
How can one update the kernel (well, I know we need to rebase drivers on top of the desired point release, but unfortunately the kernel has been added without the commit history here)? Could linux-rockchip with e.g. DTs ported be used if one needs version 6.1? I do not have a board yet, but I'm researching the opportunities.

Thank you!

dsseng, 2025-05-20:
> Hello, if you need the relevant source code, you can consult Rockchip for it. The materials we have obtained only include the SDK and the publicly available documentation on the Wiki. However, the [...]

Thank you! This means I can probably try to build using another compiler (maybe even llvm) and GNU libc to make all the dependencies incompatible with uClibc work.

RKISP is camera driver (some Userspace lib for it?) and rockit is hardware video encoder/decoder, right? So perhaps our project can just disable them in the build since we use neither.

luckfox-eng29, 2025-05-21:
@plan44 If you don't have the need to use the camera and hardware encoding, the closed-source libraries in the SDK and the drivers provided in assembly form can be ignored. That is, arm-rockchip830-linux-uclibcgnueabihf- is not necessary and can be replaced with the tools you need. I have tested that directly using the cross-compilation tool of glibc (originated from Linaro, not provided by Rockchip) to compile the kernel is without any problem. The construction of the root file system does not have any restrictions because it does not require linking to the closed-source libraries based on arm-rockchip830-linux-uclibcgnueabihf-. Previously, there were customers who successfully ported Alpine (and OpenWrt, which is based on musl) to Luckfox Pico.
Regarding the re-building of cross-compilation tools, as I answered before, we don't have relevant experience and are not clear about what is needed. To enable users to utilize all the functions (mainly the camera), we currently have no plan to support other compilation tools in the SDK.

luckfox-eng29, 2025-05-21:
@dsseng Rockchip provides an SDK for each chip. It is mentioned in the Readme that our SDK is customized based on Rockchip's SDK, and the kernel is also provided by Rockchip. It is not directly incorporated into the Linux mainline or linux-rockchip. We are not sure what specific modifications were made compared to linux-rockchip.
Using other compilation tools for compilation is feasible. The rkisp and rockit related drivers can actually be loaded, but the software libraries for calling the drivers cannot be used. The compilation of the media and app directories can be skipped during the compilation process.

dsseng, 2025-05-21:
> [@dsseng](https://github.com/dsseng) Rockchip provides an SDK for each chip. It is mentioned in the Readme that our SDK is customized based on Rockchip's SDK, and the kernel is also provided by [...]

@luckfox-eng29 thank you! So you do not know what Luckfox team has added to the Rockchip BSP kernel? Also, which of the Pico boards' DTs should one choose for the [Core1106](https://www.waveshare.com/core1106.htm?sku=26886), which wireless does it use (perhaps we'll have to add a DT node for the built-in wireless in accordance to the schematic of the board? I could not find neither any mention of Core1106 in this SDK nor a different SDK for it specifically, but believe another RV1106G3 board should be rather compatible, except no WiFi on other boards. Is it AP6256 on the Core1106, I've seen mentions of it throughout the SDK. Thanks for your help!

luckfox-eng29, 2025-05-21:
@dsseng The current luckfox-pico SDK has been modified based on the Rockchip SDK. The kernel in the Rockchip SDK and the version in the [linux-rockchip](https://github.com/rockchip-linux/kernel/) repository are different. We usually use the kernel in the SDK and do not separately pull from [linux-rockchip](https://github.com/rockchip-linux/kernel/). We have not compared the differences between the two versions.
Our modifications to the kernel mainly involve adding device tree files and defconfig files. We will only modify the source code when there are problems with the peripheral devices during adaptation. I'm not sure what you mean by Rockchip BSP kernel?
The configuration for Core1106 can be referred to the configuration of Luckfox Pico Pi. Since the peripheral circuit of Core1106 is designed by the customer, we cannot provide a separate configuration item.

dsseng, 2025-05-21:
@luckfox-eng29 thank you! Yes, thought the same, Pico Pi should be the optimal way for me to design a DT for our board. What is the wireless chip on board so we can include it in the DeviceTree? Or can we expect it to just get detected by in-kernel drivers based on its SDIO IDs? I've mostly seen those explicitly listed with a compatible to denote the chip name, but maybe that works even without it.

Thanks, dt and defconfig are easy to port atop linux-rockchip in case we need 6.1 kernel, they have rv1106 drivers in the Rockchip kernel. BSP means board support package, and perhaps there are two: one Rockchip provided you with (perhaps on the order) and the one on their GitHub I've found (has 6.1 kernel). Could you please share what fixes have you needed to apply on top of the kernel you used?

Thank you for your help!

luckfox-eng29, 2025-05-21:
@dsseng The model of the wifi chip is aic8800dc. The related source code is not directly placed in the kernel. It is located in the directory $SDK/sysdrv/drv_ko/wifi/aic8800dc.
The kernel in this repository has been modified multiple times after being updated to 5.10.160. If you need to, you can compare the commits to determine what changes were made. I am unable to list them all.

dsseng, 2025-05-21:
Thank you @luckfox-eng29! Yes, if needed I could take a look at the changes after the initial commit, I'm mostly asking what's different from the Rockchip kernel in the initial commit when it got fully imported without prior history.

I see, wifi driver is built by Buildroot as an out-of-tree module. As I haven't checked yet do you by any chance know if it can be an access point, and access point and client simultaneously? We need at least AP function for the device

plan44, 2025-05-21:
@luckfox-eng29 Thanks for the clarification. That's all I need, some context info 😄 
I understand that what luckfox can do is limited by what rockchip provides in their SDK.

I made a lot of progress with OpenWrt based on @pengphei's [efforts](https://github.com/ticklab/openwrt/tree/rockchip-22.03) yesterday, I got it built and booting ok on the 86Panel. [p44-xx-open](https://github.com/plan44/p44-xx-open) basically works! It's still WIP of course but I am pushing ongoing progress to [my fork](https://github.com/plan44/openwrt/tree/p44-rockchip-rebased-may25), and [here for the kernel](https://github.com/plan44/rockchip-kernel/tree/openwrt-p44-5.10)

The openwrt toolchain builds fine on aarch64 host, so I will use that for now and see how far I get. My next target is getting the 86panel's display to work with lvgl.

Btw: is there a matrix channel or something similar to discuss things around luckfox?

luckfox-eng29, 2025-05-21:
@dsseng I don't have any more information to share regarding the kernel version. The SDK (including the kernel) used during the initial import (including the kernel) was provided by Rockchip and had no version information. Therefore, we are not clear about the differences between the kernel in the SDK and the Rockchip kernel. All the modifications we made have been committed in this repository. For more detailed information, please consult Rockchip directly.
The configuration of the AP mode can be referred to in the forum [forums.luckfox.com](https://forums.luckfox.com/viewtopic.php?p=4614&hilit=WIFI+AP#p4614)

luckfox-eng29, 2025-05-21:
@plan44 Congrats on the progress with your project! 
We don’t have a Matrix channel at the moment, but you’re very welcome to join our [LuckFox forum](https://forums.luckfox.com/) — it’s a place to discuss issues, share your work, or ask questions.

tmakumire, 2025-12-10:
That's great to hear @plan44. Luckily I have the RV1106 so I guess I'll be using the rkdeveloptool