LuckfoxTECH/luckfox-pico #279: Source code for 86UI_demo?

plan44, 2025-05-17:
Is there source code available for the `86UI_demo` program that was added in #274 (at `project/cfg/BoardConfig_IPC/overlay/overlay-luckfox-buildroot-86panel/usr/bin/86UI_demo`) as a binary only?

I am very interested to use the 86 box, and manged to build the image after using a x86 host (tried aarch64 first, see #278). The UI86_demo sample would be a good starting point to see how to set up lvgl, sound etc.

luckfox-eng29, 2025-05-19:
Hello, the source code of 86UI_demo will be organized and uploaded to another repository. We will announce it on the wiki once the relevant documentation is completed.

plan44, 2025-05-20:
Great to hear, looking forward to have a look at that.

The most important part for me would be the lvgl driver setup - I have existing lvgl apps I'd like to port but figuring out the details of the hardware is difficult without a sample. Any chance you could share `lv_drv_conf.h` and `lv_conf.h` ahead of time?

luckfox-eng29, 2025-05-21:
The source code has been published to [luckfox_pico_86panel_ui_demo](https://github.com/LuckfoxTECH/luckfox_pico_86panel_ui_demo)

plan44, 2025-05-21:
@luckfox-eng29 great news! thank you very much!