LuckfoxTECH/luckfox-pico #281: Fail to build rootfs year 2038 issue

mhanuel26, 2025-05-20:
Hi,

I am not entirely sure which option I have enable that produce this build failure when building rootfs using ./build.sh rootfs

[image]

I was looking at ways to disable the check for 2028 year issue with 32 bit library using Buildroot without success.

How can I fix this ?

Thanks for reading,

luckfox-eng29, 2025-05-21:
Hello, the tar command has been added when building busybox in buildroot. When compiling BR2_PACKAGE_TAR, since the arm-rockchip830-linux-uclibcgnueabihf- is a cross-compilation tool based on uclibc and Y2028 is not enabled. The cross-compilation tool is provided by Rockchip and we will not make any configuration changes.

mhanuel26, 2025-05-21:
Hello @luckfox-eng29 ,

I understand, do you know of any existing workaround for this? 

Thank you,

luckfox-eng29, 2025-05-21:
Execute "./build.sh buildrootconfig" to search for "BR2_PACKAGE_TAR" and disable it. The tar command already supports this configuration by default in the compiled image.

mhanuel26, 2025-05-21:
Hi @luckfox-eng29, thank you for the info, I have an issue ading my kernel correctly, would you mind to advice, the issue is:

The kernel version is apperantly 5.10.160 (this is what uname -r shows), the default image was not having the /lib/modules directory, so I went and enable it doing

[image]

But now I have sources for 6.1.57

[image]

Would you mind to advice what is correct way to fix this issue?

Thank you very much,

luckfox-eng29, 2025-05-21:
The compilation of the kernel and the root file system in our provided SDK is separate. The buildroot is mainly used for adding software package designs to the kernel compilation process, and all operations such as this are ineffective. If you need /lib/modules/5.10.160, you should upload $SDK/sysdrv/source/objs_kernel/drv_ko/lib/modules/5.10.160 to the board after the kernel compilation. By default, we only support the insmod loading method for .ko files.

luckfox-eng29, 2025-05-21:
The latest kernel version is 5.10.160. However, when buildroot builds the image, it will not include the files in sysdrv/source/objs_kernel/drv_ko/lib/modules/5.10.160 in the build process. You can write a script to perform the copy operation of /lib/modules before packaging rootfs.img. For the specific implementation, please refer to the source code of the SDK.

mhanuel26, 2025-05-21:
Hi @luckfox-eng29 ,

Thank you for your response,

After building (./build.sh lunch then ./build.sh) I cannot see the lib directory, what am I missing?

[image]

I will appreciate your response,
Thanks again,

luckfox-eng29, 2025-05-21:
Before packaging the image, you can refer to the method provided in [Luckfox forums](https://forums.luckfox.com/viewtopic.php?p=1132&hilit=output%2Fout%2Frootfs_uclibc_rv1106#p1132) to add files. For more details, please further understand the compilation process. I am unable to assist you in determining each step of the secondary development.

closed 2025-05-21 by mhanuel26