LuckfoxTECH/luckfox-pico #282: Buildroot on core1106, wlan0 looses connection

vjoco, 2025-05-20:
I use buildroot for ultra w on a cc1106 module. After 1-2 minutes wlan0 is losing connection. What can I do?

´´´
killall wpa_supplicant
[root@luckfox root]# wpa_supplicant -B -i wlan0 -c /etc/wpa_supplicant.conf
Successfully initialized wpa_supplicant
[root@luckfox root]# udhcpc -i wlan0
udhcpc: started, v1.36.1
udhcpc: broadcasting discover
udhcpc: broadcasting select for 10.99.2.69, server 10.99.15.254
udhcpc: lease of 10.99.2.69 obtained from 10.99.15.254, lease time 28800
deleting routers
adding dns 9.9.9.9
adding dns 1.1.1.1
[root@luckfox root]# ping 8.8.8.8 -I wlan0
PING 8.8.8.8 (8.8.8.8): 56 data bytes
64 bytes from 8.8.8.8: seq=0 ttl=117 time=17.167 ms
64 bytes from 8.8.8.8: seq=1 ttl=117 time=17.671 ms
64 bytes from 8.8.8.8: seq=2 ttl=117 time=41.740 ms
64 bytes from 8.8.8.8: seq=3 ttl=117 time=73.082 ms
64 bytes from 8.8.8.8: seq=4 ttl=117 time=16.333 ms
^C
--- 8.8.8.8 ping statistics ---
22 packets transmitted, 5 packets received, 77% packet loss
round-trip min/avg/max = 16.333/33.198/73.082 ms
´´´

luckfox-eng29, 2025-05-21:
Hello, the network environment is relatively stable. There are various reasons for packet loss. I haven't encountered similar issues on the Luckfox Pico Pi W (based on Core1106). You can try using the Core1106 to ping the router's IP address and compare it with the ping test using other devices in the same environment to determine if there is any problem with the hardware.

vjoco, 2025-05-21:
I think it's solved. It is more related with the wifi access point (a lot of connected clients).

closed 2025-05-21 by vjoco