LuckfoxTECH/luckfox-pico #283: I2s sound on pico ultra

klosowskiMikolaj, 2025-05-21:
Hi 👋 
I want to drive a bigger speaker (in a waterproof enclosure, the 1W is not loud enough), so I started looking at the alternatives, one attempt at direct amplification of the speaker out using TDA2030A powered with 12V ended in magic smoke - board still works but completely no audio on the smaller speaker.
According to the https://files.luckfox.com/wiki/Luckfox-Pico/PDF/Luckfox%20Pico%20Ultra%20W.pdf
There are a couple of pins like:

- I2S0_SCLK GPIO2_A0_d -> 64
- I2S0_LRCK GPIO2_A1_d -> 65
- I2S0_MCLK GPIO2_A2_d -> 66
- I2S0_SDO3_SDI1 GPIO2_A3_d -> 67
- I2S0_SDO0 GPIO2_A4_d -> 68
- I2S0_SDI0 GPIO2_A5_d -> 69
- I2S0_SDO2_SDI2 GPIO2_A6_d -> 70
- I2S0_SDO1_SDI3 GPIO2_A7_d -> 71

That can be used as I2S. I did not find LineOut directly on the pin headers, sadly.
So I would like to connect something like https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp

Is this the correct connection?
MAX98357 BCLK ←→ I2S0_SCLK GPIO2_A0_d (Pin 64)
MAX98357 LRC ←→ I2S0_LRCK GPIO2_A1_d (Pin 65)
MAX98357 DIN ←→ I2S0_SDO0 GPIO2_A4_d (Pin 68)

then what would I need to change in https://github.com/LuckfoxTECH/luckfox-pico/blob/main/sysdrv/source/kernel/arch/arm/boot/dts/rv1106-luckfox-pico-ultra-ipc.dtsi to play normal audio via aplay/madplay?

luckfox-eng29, 2025-05-22:
If the MCLK is provided with a crystal oscillator as the clock source, your connection is fine. When rv1106 uses the I2S pins to drive the sound card, you need to set the status of the acodec_sound node and the acodec node to disabled. I haven't adapted the MAX98357. For detailed device tree configuration, please refer to the driver source code or the device tree configuration of other chip platforms.

klosowskiMikolaj, 2025-05-22:
Thank you for the response. Did you adapt any other devices for sound over I2S (then I could just buy it 😄  )?