LuckfoxTECH/luckfox-pico #284: How to correctly build the kernel modules (/lib/modules) for Linux 5.10.160 luckfox pico pi.

mhanuel26, 2025-05-21:
I am trying to build the kernel modules using the options available when running ./build.sh buildrootconfig

For instance I have selected

[image]

My target options are

[image]

When deplyoed on the target I am getting

[image]

I have check the file on the target after scp back to the host such as 

[image]

By default  the modprobe command  will not work without the /lib/modules directory, the way I found to do it so far and it generates my kernel modules structure according to what I have selected (such as the libcomposite driver) seems correct but the file format of the compiled file is incorrect. 

Can someone spot what am I missing or if there is a different approach?

Thank you for reading,

luckfox-eng29, 2025-05-22:
If you only want to load the .ko file, it is recommended to use the insmod command to directly install the driver. For rv1106, since some closed-source drivers of Rockchip have requirements for loading sequence, the /oem/usr/ko/insmod_ko.sh script is used to complete the driver loading steps. The .ko generated by the kernel compilation is ultimately placed in the /oem/usr/ko directory on the board side.

mhanuel26, 2025-05-22:
Hi @luckfox-eng29 ,

SInce my problem is with compiled file format and not the command used to insert the module into the kernel, the expected behaviour of insmod is similar to modprobe

[image]

Do you know how to correctly configure the builder for kernel sources using the buildrootconfig tool?

This is holding me a little bit to continue development, specially I need to confirm the build proecss is good enough to choose the module for making a PCBA based on core1106, so far I am not able to correcty insert a kernelmodule build with the tools, such as the libcomposite, any ideas? does someone has similar need and have found a way to build kernel sources correctly?

Regards,