LuckfoxTECH/luckfox-pico #296: usb gadget configfs can not hot-reconfigure

KaliAssistant, 2025-06-25:
I have a pico max, running ubuntu, and i'm using  [linux-usb-gadgets/gt](https://github.com/linux-usb-gadgets/gt) for my usb test, but when i want to  reconfigure the function, if i remove the configFS settings, the system will hangup. i only can disable/enable the configured settings (boot up)

```bash
ssh pico@10.71.74.75   
Warning: Permanently added '10.71.74.75' (ED25519) to the list of known hosts.
pico@foxjack:~$ ls /sys/kernel/config/usb_gadget/
g1
pico@foxjack:~$ cat rpi-sdr-tx.scheme 
attrs :
{
    bcdUSB = 0x200;
    idVendor = 0x1D6B;
    idProduct = 0x4C5B;
};
os_descs :
{
    use = 1;
    qw_sign = "MSFT100";
    b_vendor_code = 0xcd;
};

strings = (
    {
        lang = 0x409;
        manufacturer = "KALIASSISTANT";
        product = "JUST A TEST (trust me)";
        serialnumber = "1145141919810AAAAC3DE796446539A5";
    }
);
functions :
{
    rndis_usb0 : 
    {
        instance = "usb0";
        type = "rndis";
        os_descs = (
            {
                interface = "rndis";
                compatible_id = "RNDIS";
                sub_compatible_id = "5162001";
            });
        attrs :
        {
            dev_addr = "42:61:64:55:53:42";
            host_addr = "48:6f:73:74:50:43";
        };
    };
};

configs = (
    {
        id = 1;
        name = "c";
        attrs :
        {
            bmAttributes = 0x80;
            bMaxPower = 250;
        };
        functions = (
            {
                name = "rndis.usb0";
                function = "rndis_usb0";
            });
    } );
```

```bash
pico@foxjack:~$ sudo gt disable g1
[sudo] password for pico: 
pico@foxjack:~$ sudo gt enable g1
[... 421 lines cut ...]
[21139.922536] usb 3-1: Device not responding to setup address.
[21140.126471] usb 3-1: Device not responding to setup address.
[21140.330380] usb 3-1: device not accepting address 69, error -71
[21140.450380] usb 3-1: new high-speed USB device number 70 using xhci_hcd
[21140.450475] usb 3-1: Device not responding to setup address.
[21140.654626] usb 3-1: Device not responding to setup address.
[21140.862370] usb 3-1: device not accepting address 70, error -71
[21140.862441] usb usb3-port1: unable to enumerate USB device
[21226.491308] usb 3-1: new high-speed USB device number 71 using xhci_hcd
[21231.767475] usb 3-1: device descriptor read/64, error -110
[21247.383585] usb 3-1: device descriptor read/64, error -110
[21247.611711] usb 3-1: new high-speed USB device number 72 using xhci_hcd
[21247.731726] usb 3-1: device descriptor read/64, error -71
[21247.959738] usb 3-1: device descriptor read/64, error -71
[21248.067722] usb usb3-port1: attempt power cycle
[21248.463572] usb 3-1: new high-speed USB device number 73 using xhci_hcd
[21248.463663] usb 3-1: Device not responding to setup address.
[21248.667676] usb 3-1: Device not responding to setup address.
[21248.875569] usb 3-1: device not accepting address 73, error -71
[21248.995587] usb 3-1: new high-speed USB device number 74 using xhci_hcd
[21248.995722] usb 3-1: Device not responding to setup address.
[21249.199696] usb 3-1: Device not responding to setup address.
[21249.403624] usb 3-1: device not accepting address 74, error -71
[21249.403763] usb usb3-port1: unable to enumerate USB device
```

luckfox-eng29, 2025-10-27:
I conducted the test on Buildroot. The hot-reconfiguration of USB GADGET caused the kernel to crash and the system to become unresponsive. Finally, I determined that it was a bug in the dwc3 controller that caused this issue. This is the modification I made and I hope it will be helpful for you to solve the problem.

[dwc3_gadget_fix.patch](https://github.com/user-attachments/files/23164954/dwc3_gadget_fix.patch)

KaliAssistant, 2026-09-12:
[luckfox_usb.log](https://github.com/user-attachments/files/32140826/luckfox_usb.log)
@lsby @luckfox-eng29 i think [rockchip upstream 5.10](https://github.com/rockchip-linux/kernel/tree/develop-5.10) fix my problem but i just rsync the upstream kernel to sysdrv/source/kernel
@luckfox-eng29 can u take a look the upstream? i need some1 to review the diff
cuz my git changes like 9k+ modified, i've no time to review ts and make the patch file

@lsby i also tried yours, but same. i think the problem is the unbind / remove function in dwc3 or configfs
and now rockchip upstream fix my problem

KaliAssistant, 2026-09-12:
but i still dont understand why rockchip use android kernel as the sdk
and many android sh*t thing in the kernel driver code
still looking the full function 6.11+ sdk for yocto (~impossible cuz its rockchip~)

closed 2026-09-19 by KaliAssistant