LuckfoxTECH/luckfox-pico #299: PLeAsE sUpPOrT gLiBc

KaliAssistant, 2025-07-05:
i m using buildroot, python need cryptography
i try to build whl for cryptography
but need rust
i cross comp the rust for arm , cryptography  just need maturin
when i cross build maturin for rockchip ucLibc:
```
cargo install maturin --version 1.8.0 --locked                                                                                                         ─╯

    Updating crates.io index
  Installing maturin v1.8.0
    Updating crates.io index
warning: package `crossbeam-channel v0.5.13` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
warning: package `zip v2.1.6` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
    Updating crates.io index
  Downloaded tracing-serde v0.1.3
  Downloaded thiserror v1.0.64
  Downloaded thiserror-impl v2.0.3
  Downloaded toml_datetime v0.6.6
  Downloaded serde_spanned v0.6.6
  Downloaded cargo-xwin v0.18.3
  Downloaded thiserror-impl v1.0.64
  Downloaded bytes v1.7.1
  Downloaded cargo-zigbuild v0.19.7
  Downloaded log v0.4.21
  Downloaded pin-project-lite v0.2.14
  Downloaded once_cell v1.19.0
  Downloaded spin v0.9.8
  Downloaded thiserror v2.0.3
  Downloaded quote v1.0.36
  Downloaded bzip2 v0.4.4
  Downloaded tracing-attributes v0.1.27
  Downloaded target-lexicon v0.13.0
  Downloaded platform-info v2.0.3
  Downloaded toml v0.8.15
  Downloaded proc-macro2 v1.0.85
  Downloaded indicatif v0.17.8
  Downloaded serde v1.0.216
  Downloaded cargo-config2 v0.1.30
  Downloaded tracing-core v0.1.32
  Downloaded cc v1.1.21
  Downloaded serde_derive v1.0.216
  Downloaded tracing v0.1.40
  Downloaded indexmap v2.5.0
  Downloaded crossbeam-channel v0.5.13
  Downloaded toml_edit v0.22.16
  Downloaded zip v2.1.6
  Downloaded serde_json v1.0.133
  Downloaded memchr v2.7.2
  Downloaded zerocopy v0.7.34
  Downloaded winnow v0.6.13
  Downloaded regex v1.10.6
  Downloaded rustix v0.38.34
  Downloaded regex-syntax v0.8.4
  Downloaded regex-automata v0.4.7
  Downloaded bzip2-sys v0.1.11+1.0.8
  Downloaded libc v0.2.158
  Downloaded ring v0.17.8
  Downloaded 43 crates (8.8MiB) in 1.01s (largest was `ring` at 4.0MiB)
   Compiling proc-macro2 v1.0.85
   Compiling unicode-ident v1.0.12
   Compiling cfg-if v1.0.0
   Compiling libc v0.2.158
   Compiling version_check v0.9.4
   Compiling once_cell v1.19.0
   Compiling shlex v1.3.0
   Compiling serde v1.0.216
   Compiling memchr v2.7.2
   Compiling rustix v0.38.34
   Compiling linux-raw-sys v0.4.14
   Compiling bitflags v2.5.0
[... 263 lines cut ...]
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: /home/kali/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/armv7-unknown-linux-gnueabihf/lib/libstd-2d46c80c4783e43e.rlib(std-2d46c80c4783e43e.std.5e38d238f43fcd69-cgu.0.rcgu.o): in function `<std::sys::process::unix::unix::<impl std::sys::process::unix::common::Command>::posix_spawn::PosixSpawnattr as core::ops::drop::Drop>::drop':
          /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/std/src/sys/process/unix/unix.rs:666: undefined reference to `posix_spawnattr_destroy'
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: /home/kali/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/armv7-unknown-linux-gnueabihf/lib/libstd-2d46c80c4783e43e.rlib(std-2d46c80c4783e43e.std.5e38d238f43fcd69-cgu.0.rcgu.o): in function `std::sys::process::unix::unix::<impl std::sys::process::unix::common::Command>::posix_spawn':
          /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/std/src/sys/process/unix/unix.rs:727: undefined reference to `posix_spawnattr_setsigdefault'
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: /home/kali/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/armv7-unknown-linux-gnueabihf/lib/libstd-2d46c80c4783e43e.rlib(std-2d46c80c4783e43e.std.5e38d238f43fcd69-cgu.0.rcgu.o): in function `<std::sys::process::unix::unix::<impl std::sys::process::unix::common::Command>::posix_spawn::PosixSpawnFileActions as core::ops::drop::Drop>::drop':
          /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/std/src/sys/process/unix/unix.rs:656: undefined reference to `posix_spawn_file_actions_destroy'
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: /home/kali/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/armv7-unknown-linux-gnueabihf/lib/libstd-2d46c80c4783e43e.rlib(std-2d46c80c4783e43e.std.5e38d238f43fcd69-cgu.0.rcgu.o): in function `<std::sys::process::unix::unix::<impl std::sys::process::unix::common::Command>::posix_spawn::PosixSpawnattr as core::ops::drop::Drop>::drop':
          /rustc/6b00bc3880198600130e1cf62b8f8a93494488cc/library/std/src/sys/process/unix/unix.rs:666: undefined reference to `posix_spawnattr_destroy'
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: /tmp/cargo-installCvcQjg/armv7-unknown-linux-gnueabihf/release/deps/libfiletime-ba5d887c448da8ab.rlib(filetime-ba5d887c448da8ab.filetime.6b239d26ff768243-cgu.0.rcgu.o): in function `filetime::set_file_handle_times':
          filetime.6b239d26ff768243-cgu.0:(.text._ZN8filetime21set_file_handle_times17h35aeca83ae54c8ceE+0x230): undefined reference to `futimes'
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: /tmp/cargo-installCvcQjg/armv7-unknown-linux-gnueabihf/release/deps/libring-7d61a9f5da70095c.rlib(ring-7d61a9f5da70095c.ring.d4653431a8acfab1-cgu.09.rcgu.o): in function `ring::cpu::arm::init_global_shared_with_assembly':
          ring.d4653431a8acfab1-cgu.09:(.text._ZN4ring3cpu3arm32init_global_shared_with_assembly17he363bf5a773aa7e6E+0xc): undefined reference to `getauxval'
          /media/kali/KALI512GDAT/github.workdir/KaliAssistant__Fox-Jack/.kaliworkingspace/luckfox-pico/tools/linux/toolchain/arm-rockchip830-linux-uclibcgnueabihf/bin/../lib/gcc/arm-rockchip830-linux-uclibcgnueabihf/8.3.0/../../../../arm-rockchip830-linux-uclibcgnueabihf/bin/ld.bfd: ring.d4653431a8acfab1-cgu.09:(.text._ZN4ring3cpu3arm32init_global_shared_with_assembly17he363bf5a773aa7e6E+0x20): undefined reference to `getauxval'
          collect2: error: ld returned 1 exit status
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their path specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://doc.rust-lang.org/cargo/reference/build-scripts.html#rustc-link-lib)

error: could not compile `maturin` (bin "maturin") due to 1 previous error
error: failed to compile `maturin v1.8.0`, intermediate artifacts can be found at `/tmp/cargo-installCvcQjg`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
```

😭😭😭

KaliAssistant, 2025-07-05:
cross compile everything, but everything need glibc
f**king rockchip

chhartmann, 2025-07-06:
Already some time ago, but you could have a look here: https://github.com/chhartmann/luckfox_pico_rust. Nine months ago the standard library in the luckfox image was too old for rust.

AliMoal, 2025-07-10:
> [@KaliAssistant](https://github.com/KaliAssistant) just use yocto. its not that hard to adapt existing meta-rockchip to support rv1103/rv1106 cpus.

I'm new to yocto, could you show me the way to use it for luckfox boards? Buildroot is not a good option at all especially when you want to compile a new program or add any new packages. Unfortunately, Ubuntu is not support anymore. Luckfox company left us alone :)

Maobuff, 2025-07-12:
@AliMoal Did you check what packages yocto provides? nm is defiantly there. Almost every package that you can imagine is there. Even if package is missing you can create your own meta layer and include that package.

AliMoal, 2025-07-12:
> [@AliMoal](https://github.com/AliMoal) Did you check what packages yocto provides? nm is defiantly there. Almost every package that you can imagine is there. Even if package is missing you can create your own meta layer and include that package.
> 
> Prebuild packages for embedded are evil to be honest. You will encounter library versions mismatch, packages build with not the right options or options enabled that you dont want at all. You mentioned word "minimal" previously. prebuild and minimal are not even close together.

Tbh, working with yocto is too hard, thats why i want to choose the easier way :)