LuckfoxTECH/luckfox-pico #315: Request for english response to: USB port next to ethernet on Pico Ultra W not working #131

danielw1990, 2025-09-21:
What was the solution in issue 131?
https://github.com/LuckfoxTECH/luckfox-pico/issues/131

Google translate Chinese to English only produces word salad.

And the latest Ubuntu images still seems to have the same issue.

x-magic, 2025-11-29:
The general idea is that the USB Type-A host and USB Type-C host are actually wired together. Essentially the same logic port. You can't use them at the same time. The Type-A port only works if there's nothing plugged into the Type-C port. Not even power. To use Type-A port, you'll need to power the board through GPIO pins or PoE module. 

Also the USB mode must be set to host mode. Otherwise the USB port will act as a client device. `peripheral` mode is for adb, and `otg` mode... TBH I have no idea what it means or use for. It's confusing.

danielw1990, 2025-11-29:
Thanks!

I do know have some experience with other Luckfox devices.
And adb actually means you can use Android Debug Tools

For example to shell login via USB connection.

example:  
```
//list adb devices
adb devices -l
//connect to one without <>
adb connect <device_id_from_list>
//open shell to connected device
adb shell 
```
Then you get interactivity similar to a ssh connection. But not via IP networking. Not to be confused with the devices ability to do IP networking via USB. Which can be used for regular ssh if such a service is installed and running. This can even be used to change network settings if it is completely unconfigured/incompatible with your network. And a reload of networking configs will not break the connection.

I basically used it as a more modern alternative to TTY. With USB speeds and official hot-plug support.